cornellev/component_chassis #22: Investigate Better Forged Mold Designs

lucaslibshutz, 2025-09-05:
As our first attempt at forged carbon fiber went well, the next step is to see what better methods of manufacturing molds there are to use in the future, possibly for a component like the mirrors, or whatever the team might need.

@beefserky12 fill out dates when you can, ping me when you've added them.

beefserky12, 2025-09-13:
https://www.easycomposites.co.uk/learning/compression-moulding-forged-carbon-fibre

beefserky12, 2025-09-13:
# 3D Printed Molds 

## Process

- A mold is designed in CAD and produced using a 3D printer (commonly PLA, ABS, or PETG).
- The mold surface is post-processed (sanded, sealed, and coated with release agents or epoxy surfacing coat) to prevent resin leaks and improve finish.
- Chopped carbon tow is combined with resin, pressed into the mold, and cured either at room temperature or in an oven.

## Benefits

- Low cost: Inexpensive compared to CNC machining or cast resin molds.
- Fast turnaround: Ideal for prototyping and one-off parts, since molds can be printed quickly.
- Complex geometry: 3D printing enables intricate designs and organic shapes without expensive tooling.

## Drawbacks

- Surface quality: Print layer lines and porosity require sealing and finishing
- Thermal resistance: Thermoplastics can deform under heat and pressure during curing.
- Durability: Limited lifespan; molds wear down quickly with repeated use.

# Epoxy Resin Cast Molds

## Process

- A positive master model is created (via CNC machining or 3D printing).
- The master is coated in a release agent, and epoxy tooling resin is poured over it to form a negative mold.
- Once cured, the resin mold is sanded and polished, then used for layup of chopped carbon tow and resin.

## Benefits

- Improved durability: Stronger than 3D printed molds, capable of moderate thermal cycles.
- Better surface finish: Epoxy resins produce smoother mold surfaces, leading to higher-quality part finishes.
- Repeatability: Suitable for small batch production (multiple identical parts).

## Drawbacks

- Time-consuming: Requires creation of a master, casting, and curing before use.
- Higher cost: Epoxy tooling materials are more expensive than basic 3D prints.
- Limited lifespan: Still less durable than metal molds, especially under high pressure/temperature cycles

# CNC-Machined Aluminum or Steel Molds

## Process

- The mold is designed in CAD and CNC-machined directly into aluminum or steel.
- Precision machining allows for highly accurate surfaces and fine details.
- The mold can be heated during curing and withstand compression molding pressures, ensuring a high-quality final product.

## Benefits

- High durability: Can withstand repeated thermal cycles and high pressures.
- Excellent finish: Produces professional-grade surfaces with minimal post-processing.
- Structural reliability: Best option for producing functional, load-bearing carbon fiber parts.
- Long-term investment: Useful if the team plans to make multiple iterations or larger quantities.

## Drawbacks

- Cost: Significantly more expensive in terms of material and machining time.
- Lead time: CNC programming and machining can take longer than 3D printing.
- Overkill for one-off parts: If parts are not repeated, the investment may not be justified.
-

lucaslibshutz, 2025-09-22:
@beefserky12 @zachfeldman3 update dates when you can

lucaslibshutz, 2025-10-16:
Looks good, closing

closed 2025-10-16 by lucaslibshutz