cornellev/component_chassis #52: Rough Design of Hinge for PDR

ei-thirikyaw, 2025-09-13:


lucaslibshutz, 2025-09-15:
@ei-thirikyaw How are you looking for this week? Do you need any help from me or @zachfeldman3 ?

zachfeldman3, 2025-09-15:
Yeah feel free to reach out if you need anything or have any questions or want any help @ei-thirikyaw

ei-thirikyaw, 2025-09-18:
[image]

[image]

Hinge + Hinge Pin
Attachment Point Front Side of the Chassis 
Material: PVC 
Manufacturing Method: 3D Printing
Dimensions: Height 9 cm
Bolt: M6 (Clearance Hole size 6.3 mm)

lucaslibshutz, 2025-09-21:
We got a design and feedback! Closing.

closed 2025-09-21 by lucaslibshutz