cornellev/component_chassis #53: CFD Course Module: Foundations of CFD

lucaslibshutz, 2025-09-13:
Reporting all notes from this module here

zachfeldman3, 2025-09-15:
Ok I'll take a look

lucaslibshutz, 2025-09-17:
@zachfeldman3 any thoughts?

zachfeldman3, 2025-09-17:
Sorry- have been cooked with fluids. Will document all my thoughts in the morning.

zachfeldman3, 2025-09-17:
I read what you wrote. I think it's a good start, and I definitely need to do more research into the mathematical model / governing equation because I'm not yet perfectly knowledgeable on the fundamental laws of fluid flow. I disagree in the fact that hand calcs are necessary. I mean, usually they are, but for our purposes, unless if we want to approximate our car to a sphere, our geometry is way too difficult to approximate, so I think it's best to just compare our results to our previous cars to see if we're in the realm of sanity

lucaslibshutz, 2025-09-17:
@zachfeldman3 this module is just foundations - do you think its enough? For mathematical model thats a separate issue

lucaslibshutz, 2025-09-18:
I think we're good on this, tbh. I'll close this, and if we need to add more we can. Let's focus on getting the big idea on CFD before getting specific on documenting specific modules.