cornellev/component_chassis #55: Continue researching on carbon fiber structure and stress/strain response and familiarize with what ACP actually is and what it does

ylin1004, 2025-09-16:


ylin1004, 2025-09-17:
### The Load Bearing Mechanism of Carbon Fiber:

- Fibers carry the main load. 
Due to their much higher strength and elastic modulus, the reinforcing fibers are the primary load-bearing elements in a composite, particularly when the force is applied along the fiber's direction. The matrix, being less stiff, supports a negligible portion of the tensile load.

- The matrix transfers the load. 
The matrix is essential for distributing the external stress and transferring it to the fibers through the fiber-matrix interface (IFSS).  In a continuous fiber composite, the fibers usually carry most of the load, since the modulus of the fibers is much higher than that of the matrix. As a consequence the term 'load transfer' is conventionally used to refer to that from matrix to fiber. However, in this paper, load transfer from fiber to matrix (i.e. fraction of the load applied to the fiber that is transferred to the matrix) is addressed. **The stronger the bond between fiber and matrix, the more effective is the load transfer in either direction**.

- The load is shared based on material properties. 
The division of the total load between the fiber and matrix depends on their respective stiffness and volume fraction, as described by the rule of mixtures for composite materials. An increase in the fiber volume fraction or the fiber-to-matrix stiffness ratio causes more stress to be transferred to the fibers.

lucaslibshutz, 2025-09-17:
> 1. Consider Geometric Variations and analyze each one

@ylin1004 maybe you could expand more on what you mean by "geometric variations" - is that parts of the baseplate that could be incompatible with the ACP concepts you've outlined?

Also, just something to know, now there is no ACP "Post" tool - all post analysis is handled inside of ANSYS Mechanical with an ACP helper tool inside of the app. If you need some more resources on that I can post them here. 

Great work so far though!! 🚀

ylin1004, 2025-09-21:
I'm going to change the dates for this on. Busy week with 2 prelims and other work.

ylin1004, 2025-09-27:
Working on next practices such as T-joint and try to figure out equations for failure criterion

closed 2025-09-27 by ylin1004