cornellev/component_chassis #68: Researching New Manufacturing Techniques

ei-thirikyaw, 2025-09-29:


lucaslibshutz, 2025-10-03:
@ei-thirikyaw move all of your notes from the milestone in here, will be easier to keep track of that way

ei-thirikyaw, 2025-10-04:
**Mold preparation**

Dry Guide Coat
Using a dry guide coat to make cracks and scratches more pronounced will ensure we are not oversanding or sanding the wrong areas. A dry guide coat is recommended over a wet guide coat because with wet guide coats it can sometimes cover up the crevices because it’s in a can and it’s hard to apply to coat evenly across the surface. When the dry guide coat is applied over the surface, it will sink into the areas with scratches and crevices. We can sand those areas until the guide coat is no longer visible. 

Flap Wheels
When we are assembling molds, a lot of the uneven surfaces are where the molds attach. We often have to apply bondo to seal the cracks or sand when the parts are not aligned. However, the pneumatic sander cannot reach those small edges. We can consider using a flapwheel for those areas. You can just attach it to a drill so we don’t need to buy new machinery. They’re very low cost. [You can buy many for just $30](https://www.amazon.com/s?k=flap+wheels&i=industrial&hvadid=616990991663&hvdev=c&hvexpln=0&hvlocphy=9194752&hvnetw=g&hvocijid=14203578008975040299--&hvqmt=e&hvrand=14203578008975040299&hvtargid=kwd-39786290&hydadcr=24664_13611849&mcid=6037ca9687123b9081359b23d4e445fd&tag=googhydr-20&ref=pd_sl_7ivmfzl0pp_e).

Increasing Visibility When Moving Up High Grits
Magnifying glasses
Flashlights

Mold Creation (Examples from Others)

[AJ Hartman Aero](https://www.youtube.com/watch?v=39NvpxUCZLo) (Vacuum Infusion):
(Dry Sanding w/ Pneumatic Sander): 320 → 600 → 1000 → 1500 → Polishing Pad  
(Wet Sanding with Hand): 1000 Grit  → Polishing Pad with Pneumatic Sander

[Easy Composites](https://www.youtube.com/watch?v=KpJdwryFj6k):
(Dry Sanding) 240 Grit to help bond with resin  → Coat with Epoxy Coating Resin EP-XCR → Coat with XCR again (We use Duratech instead of Resin that’s ok) → Wet Sanding 800 Grit → Wet Sanding 1200 Grit → Polished by buffing with NW1 Polishing Compound 

**Curing**
Currently we don’t use any heating method to speed up the vacuum infusion process. Curing can happen at room temperature, or mild heat can be applied with heated molds of blankets to speed up the process. Heated molds are not feasible for us since it requires building rods or water channels into the molds which would increase the costs and time. However, we can consider using heated blankets to preheat the molds or after vacuum infusing the molds for curing. The price ranges from $100 - $1000. They speed up the manufacturing process to a few hours rather than overnight. At 60C the molds will only need 2-4 hours to cure. At 80–100 °C the molds will only need 1–2 hours to cure. 
https://speautomotive.com/wp-content/uploads/2021/01/Lehman_High-Tg-Fast-Cure-Resin-System.pdf

closed 2025-10-06 by ei-thirikyaw